buu342/N64-UNFLoader #17: Ensure new EverDrive OS update didn't break anything

buu342, 2020-09-10:
If it did I will probably cry myself to sleep, no biggie.

buu342, 2020-09-10:
UNFLoader uploads fine with the 3.0 on the new OS. However, the USB library is broken. Supposedly, a magic value is implemented now, so it's a matter of putting the code in X7 mode if it detects this magic value.

buu342, 2020-09-11:
Broken on the X7 as well
[WAKE ME UP](https://external-content.duckduckgo.com/iu/?u=http%3A%2F%2Fl7.alamy.com%2Fzooms%2Fb97675b9a9dd40328247b420c2fc6a7c%2Fman-reading-bible-and-taking-drugs-d0ttnj.jpg&f=1&nofb=1)

buu342, 2020-09-12:
The samples are out. Time to have some fun:
https://github.com/krikzz/ED64/tree/master/ED64-XIO

networkfusion, 2020-09-12:
The better resource is in 😁 https://github.com/N64-tools/ED64/ (make sure you keep checking the branches)...

buu342, 2020-10-16:
Fixed in the most recent commit
https://github.com/buu342/N64-UNFLoader/commit/8c06caa08d0cfe4ff3fe5dc52a3521bffb7a1406

closed 2020-10-16 by buu342