buu342/N64-UNFLoader #28: EDX7 + macOS Testing

danbolt, 2020-11-26:
Did some tests with a more recent version of the library and samples.

I was using
1. EverDrive-64 X7 with OS 3.05
2. macOS Catalina 10.15.7 with the drivers installed as per the guide

### Example 1

### Example 1

buu342, 2020-11-26:
Fantastic! Thanks for testing :heart:

networkfusion, 2020-11-27:
@danbolt It would be great if you can take a look at the new CICD for macOS (https://github.com/buu342/N64-UNFLoader/blob/master/azure-pipelines.yml) and see if you can get it working... It is the only one that doesnt work... so currently commented out.

danbolt, 2020-11-27:
Sure thing. I might need to check with my employer first if I'm looking
into Azure, so I'll let you know if that's okay.

networkfusion, 2020-11-27:
@danbolt No need to set up your own... just create a fork and then a PR. any changes you make to the yml will automatically build against this repo. I helped @buu342 get it working 😁

networkfusion, 2020-11-27:
just as an example... 
[image]

buu342, 2020-11-27:
Reopening to make this issue is easier to find :feelsgood:

buu342, 2020-11-27:
We can also use this opportunity to flesh out the "How to build UNFLoader on Mac" section.

danbolt, 2021-02-13:
Hey team, I've set up a potential Azure pipeline here:

https://github.com/buu342/N64-UNFLoader/pull/45

@networkfusion , it seems editing `azure-pipelines.yml` hasn't updated the CI. Perhaps there's something else I need to change on my end?
[image]

danbolt, 2021-02-13:
macOS prevents running unsigned binaries by default. If a user wants to use an unsigned program for the first time, they will need to go to the **Security & Privacy** panel in the System Preferences menu and explicitly allow it to be run.

[image]

buu342, 2021-02-14:
Awesome. The Azure builds are setup, and the macOS build instructions are written, so I can go ahead and close this issue. Thanks a lot for the help Daniel :smile:

closed 2021-02-14 by buu342