buu342/N64-UNFLoader #38: [Typo Only?] Outputted help incorrectly lists the CIC types

Drakonas, 2021-02-05:
Output from the official 64Drive tool:
    -c <int>                       Set CIC emulation type (RevB/HW2 only)
               0 - 6101 (NTSC)          1 - 6102 (NTSC)
               2 - 7101 (PAL)           3 - 7102 (PAL)
               4 - x103 (All)           5 - x105 (All)
               6 - x106 (All)           7 - 5101 (NTSC)
    
    Output from your tool:
    -c <int>                       Set CIC emulation type (64Drive HW2 only).
               0 - 6101 (NTSC)         1 - 6102 (NTSC)
               2 - 7101 (NTSC)         3 - 7102 (PAL)
               4 - x101 (All)          5 - x102 (All)
               6 - x106 (All)          7 - 5101 (NTSC)

As you can see, types 4 and 5 in your tool are listed incorrectly. x105 is used for a handful of games like Legend of Zelda: Ocarina of Time.

Is your tool only forwarding the types, the same as the official tool? If so, I'd assume the command still works properly. The help output may just need to be updated.

buu342, 2021-02-05:
That is definitely a typo on my end. In fact, I've actually written the tool in a way that you were supposed to specify either the CIC exactly (ex, `-c 6102`) or just the number value (ex, `-c 1`), but quickly looking at the code, I must not have been thinking properly, because that's not what the code I wrote does at all. It only does option 1, but by sheer luck of my incompetence it didn't cause any problems (or at least no one had brought it up). Whoops!

Fixed in the latest commit 211cf9a :sweat_smile:

closed 2021-02-05 by buu342

Drakonas, 2021-02-05:
Getting your hands on a 64Drive lately is really hard, so it's to be expected nobody is probably using it. Yeah, I forgot to mention when I tried to use -c 1 it failed and said "1 is not a valid CIC" or something like that.

Drakonas, 2021-02-05:
Forgot to ask. Will it handle both 6105 and 7105 as -5 properly?

buu342, 2021-02-05:
It should now :grin:

commit 211cf9ae (2021-02-05): Fixed #38