buu342/N64-UNFLoader #39: Loading bar doesn't show (unicode incompatibility?)

Drakonas, 2021-02-05:
Is the loading bar actually being processed? I've tried various fonts and both Powershell and Command Prompt windows. The character doesn't seem to be recognized. Is there something I'm missing?

This is with the latest commit from an hour ago.

[image]

buu342, 2021-02-05:
That's interesting. Does this problem also occur with the 64Drive application? Asking because it uses the same character set for the loading bar.

Drakonas, 2021-02-06:
@buu342 Yes it does. But in the 64Drive app it actually just shows one of these characters.

buu342, 2021-02-06:
I don't think that'd be the culprit. Hmm...
I was gonna guess that it's either because your Windows was missing [the extended ASCII set](https://devblogs.microsoft.com/wp-content/uploads/sites/33/2019/02/5-Buffers-extended-ascii-codepage.png), or because your active codepage (You can check with the `chcp` command in cmd) doesn't support extended ASCII. But since you said it displays at least one of the characters correctly on the 64Drive application, that does make me wonder...

Drakonas, 2021-02-06:
FYI, chcp reports "Active code page: 65001".

> But since you said it displays at least one of the characters correctly on the 64Drive application, that does make me wonder...

No what I meant was it shows only one of the same invalid characters from the 64Drive application. It doesn't have a bar at all. It's just a single bad character that doesn't change.

[image]

Drakonas, 2021-02-06:
By the way, I patched that line and get a compiler error.

    Severity	Code	Description	                        Project	        File	                                                        Line	 Suppression State
    Error	        C1016	#if[n]def expected an identifier	UNFLoader	X:\Games\N64\64Drive\N64-UNFLoader-master\UNFLoader\helper.cpp	308

Drakonas, 2021-02-06:
This is odd. After I set it to `#ifndef i`, it still shows an invalid character. o.O

~[Screenshot 4](http://mars.whatbox.ca:42424/img/cmd_2021-02-06_16-28-24.png)

By the way the code shows, it should be printing a # character. Or is it?

The definitition is here:

`// Printing
    #define pdprint(string, color, ...) __pdprint(color, string, ##__VA_ARGS__)`

Do you have the arguments in the correct order?

Drakonas, 2021-02-06:
It seems you were correct on the codepage being wrong. I changed it to 1252, and the characters show now.

buu342, 2021-02-06:
Awesome! Glad you got it figured out :smiley:

closed 2021-02-06 by buu342