buu342/N64-UNFLoader #40: [Feature request] Detect the CIC code automatically

Drakonas, 2021-02-05:
This is very much possible, and there already exists a tool to do it:

[RN64CRC](https://www.smwcentral.net/?p=section&a=details&id=8799)

However, they didn't provide the source code, so I'm unsure how easy it would be to obtain the rights to utilize it.

I've constructed a script that utilizes the tool to automatically pull out the CIC from the 5 lines outputted, and then it forwards that to UNFLoader automatically. With the new commit allowing CIC numbers to be passed directly, it makes it much easier.

Just a thought. Technically a function could be made on one's own, as the CIC can be confirmed by calculating an MD5 hash of the bootcode inside the rom, which is located in every single N64 ROM and starts at offset 0x40 and has a size of 4032 bytes.

(see [here](https://www.retroreversing.com/n64bootcode) for more detail)

It's very possible this is the method that the RN64CRC tool uses to find what CIC chip the game has.

That page doesn't have every MD5 hash listed, but it does list what games have each CIC, so it should be easily obtainable.

Drakonas, 2021-02-05:
FYI, if someone wants my script, I'll post it. Make sure rn64crc, its required DLL's, and UNFLoader is located in %PATH% somewhere. Adding the script itself to the path also kind of helps. :)

    @echo off
    setlocal EnableDelayedExpansion
    
    setlocal disabledelayedexpansion
    set "x=%~1"
    If not defined x goto :endnoargument
    endlocal & set "x=%x:!=^!%"
    
    set x
    
    set "output_cnt=0"
    for /F "delims=" %%f in ('rn64crc.exe "!x!"') do (
        set /a output_cnt+=1
        set "output[!output_cnt!]=%%f"
    )
    set "isRomPassed=!output[2]:~0,23!"
    If "!isRomPassed!" == "No usable files found.." goto :endinvalidfile
    
    set "romCIC=!output[4]:~23,4!"
    
    echo CIC found: !romCIC!
    
    UNFLoader.exe -r "!x!" -c !romCIC! -t 3 -h 12
    
    goto :eof
    
    :endinvalidfile
    echo File is not a valid N64 rom. Exiting...
    goto :eof
    
    :endnoargument
    echo No file was passed. Exiting...
    
    :eof
    ping localhost -n 3 >NUL
    exit

buu342, 2021-02-05:
That's very interesting and I'd be glad to implement that :smile:

I'll have to ask N64brew if anyone knows the method for computing the CRC's, if there is one. I suspect there should be some sort of lookup table for this instead.

Drakonas, 2021-02-05:
I updated my script. Apparently my forced timeout wasn't registering either, due to the fact that I forgot that :eof is a hardcoded direction to the literal end of file.... You can't override that. Sometimes I hate MS-DOS/Batch. lol.

Anyway, happy to help! Thanks. :)

Drakonas, 2021-02-05:
By the way, if this is added, is there a way to detect when a rom file is passed as a sole argument? Because, in theory, this would remove all other arguments' requirements.

Doing this would make it possible for direct drag and drop of the rom onto the UNFLoader.exe. It would make it far more versatile.

buu342, 2021-02-06:
Sure, this can be looked into.

buu342, 2021-02-10:
d79303d should allow you to upload a ROM without specifying `-r`. Please check if everything's working on your end.

Drakonas, 2021-02-10:
The filepath without `-r` does work. Cool, now just for the longer part of the code. Haha.

buu342, 2021-02-12:
Added in 48052c9

closed 2021-02-12 by buu342

Drakonas, 2021-02-12:
Awesome! Thank you! And very interesting on the MD5 hashes being wrong.

Drakonas, 2021-02-23:
Only a slight find. Looks like the CIC's don't include Japanese 8### CIC's.

buu342, 2021-02-23:
Doesn't 8303 get covered by the X303 value (aka `-c 6`)?

Drakonas, 2021-02-23:
No a 303 rom was being detected as "unknown", and running with the current build it says "CIC set to 0."

buu342, 2021-02-23:
Make sure you do `-c 6`, and not `-c 8303` or `-c x303`. Otherwise it'll assume 0. I'll make the program terminate instead if the value isn't an expected one...