buu342/N64-UNFLoader #41: [Bug] timeout argument not working correctly

Drakonas, 2021-02-05:
UNFLoader.exe -r "E:\Games\N64\EverDrive 64\1 US - A-M\Lode Runner 3-D (USA).z64" -c 6102 -t 3 -h 12
Running this command instantly closes the process after it reaches 100%. No timeout occurs. Same with -t 3000, so it doesn't seem to be milliseconds either.

buu342, 2021-02-05:
Timeout is for debug mode specifically, which I know isn't very clear... 

The intended purpose for it was for unit tests or similar purposes, where a ROM is executed with debug logging for a specific period of time and then closed soon after. I can patch it to work outside of debug mode as well :slightly_smiling_face:

Drakonas, 2021-02-05:
The main use I have for it is using it in automated scripts. Leaving the window open with a "pause" at the end is kind of redundant. :P

buu342, 2021-02-06:
The timeout should (supposedly) remove that pause at the end. I'll check this code later today or tomorrow since my exams are over.

Drakonas, 2021-02-06:
Sounds good. And yes, the timeout does remove the pause at the end. It just ends instantly without any delay, regardless of the seconds given.

buu342, 2021-02-09:
Latest commit (ca7408e) should now have timeout for non debug mode usage. Please check if everything is working!

buu342, 2021-02-09:
I uhh, managed to commit the wrong file... Pushed a backup that I made before I pulled the changes from Korgeaux's PR's. The correct commit it (3d4b413) :sweat_smile:

Drakonas, 2021-02-09:
I'll give it a try asap. :)

Drakonas, 2021-02-10:
Timeout still doesn't seem to work. I tried this command in a command prompt:
    UNFLoader.exe -r "007 - GoldenEye (USA).z64" -c 6102 -t 3 -h 12

buu342, 2021-02-10:
That's weird. `CLOCKS_PER_SEC` must differ between Linux and Windows...

buu342, 2021-02-11:
Yeah after doing a bit of reading, the return value `clock()`, and by extension `CLOCKS_PER_SEC` seems to differ in a varying amount of ways for all sorts of reasons. 

tl;dr, there is no one singular function for both Windows and Linux which'll return the same value. Even on different machines running the same OS, the values of `clock()` can differ plenty.

I'm moving this to low priority, since the code is working, just not with a consistent time between devices... The workaround is to just change the value of the `-t` to something that works best for the user.

Drakonas, 2021-02-11:
https://stackoverflow.com/questions/12244153/behaviour-of-clocks-per-sec-in-different-operating-systems

buu342, 2021-02-11:
Yeah, there's a bunch of other intricacies as well with `clock()`, like how it's based on how long the program is running for, so `sleep()` screws with that.

Drakonas, 2021-02-11:
I edited my post. The example uses gettimeofday, not sleep().

buu342, 2021-02-11:
Ugh, for some reason I was thinking that `clock_gettime()` and `gettimeofday()` weren't part of standard libs like `time.h`... I'll give it a try in a bit.

Also I was referring to how *my* code uses `sleep()` (particularly in debug mode), and how that messes with the return value of `clock()` (because `clock()` returns CPU time, and `sleep()` doesn't consume any of that)

Drakonas, 2021-02-11:
Ah. I see. Well, there's no rush on this. I was just noting my findings.

buu342, 2021-02-18:
I switched to using `time()` instead of `clock()`. Can you please test 00e85a9?

Drakonas, 2021-02-19:
Well, I started it with a timer of 3 seconds. 20 minutes later it's still staying up with "Press any key to continue, or wait for timeout."

buu342, 2021-02-19:
I missed a spot... Try ad4f5b3.

Drakonas, 2021-02-19:
It now goes back to its old method of just immediately ending and not having a timeout at all.

Drakonas, 2021-02-19:
Perhaps this might help? https://www.pluralsight.com/blog/software-development/how-to-measure-execution-time-intervals-in-c--

buu342, 2021-02-19:
You've been building these yourself, right? Can you try grabbing one of the Azure builds instead and seeing if that works?
I already tested this with both debug mode on and off on:
- Windows 10
- Windows 7
- Windows XP
- Linux

And I have been unable to replicate the timeout issue. It's working perfectly on the devices I've tested...

Drakonas, 2021-02-19:
Yes I've been using Visual Studio 2019 to build. I'll give it a try later. Really strange.

buu342, 2021-02-22:
[image]

Drakonas, 2021-02-22:
Lol. Apologies. Weekends typically keep me busy. I'll take a look tomorrow during work. Heading to bed now and I've been busy all day. Haha.

Drakonas, 2021-02-23:
I tried the Azure build and it also doesn't work properly.
[image]

buu342, 2021-02-23:
Does it usually take that long to upload on your 64Drive? For instance, a 32MB ROM uploads in 2 and a half seconds on mine...

The timeout value starts counting down as soon as the application opens, so it might simply be that, by the time the upload has finished, the timeout value is long past.

Drakonas, 2021-02-24:
It is probably due to me running USB over ethernet. No data is lost, but the transmission is probably slower. I haven't had any issues. I would figure the timeout would start after it completes, as that is what I'm used to for timeout options. lol.

buu342, 2021-02-24:
That is a fair assumption, lol...
I put the timer at the start of the program as that would cover my ass if it terminated unexpectedly. I guess I won't take the lazy way out and do it properly :sweat_smile:

buu342, 2021-02-24:
Hopefully, the conclusive commit. 4dc1ffd

Drakonas, 2021-02-24:
That fixed it! Tested with a couple various commands. :)

buu342, 2021-02-24:
:partying_face:

closed 2021-02-24 by buu342