buu342/N64-UNFLoader #47: Potential undefined behaviour for 64Drive and SummerCart64 on macOS

danbolt, 2021-02-13:
**tl;dr:**
[image]

**Actual issue:**
When I compile `UNFLoader` on macOS with Apple Clang 12, I get the following compiler warnings

```
g++ -D LINUX -o UNFLoader main.cpp helper.cpp device.cpp debug.cpp device_64drive.cpp device_everdrive.cpp device_sc64.cpp Include/lodepng.cpp -lncurses -lftd2xx -lpthread -Wl,-rpath /usr/local/lib -L/usr/local/lib
device_64drive.cpp:93:22: warning: passing an object that undergoes default argument promotion to 'va_start' has undefined behavior [-Wvarargs]
    va_start(params, numparams);
                     ^
device_64drive.cpp:88:78: note: parameter of type 'u8' (aka 'unsigned char') is declared here
void device_sendcmd_64drive(ftdi_context_t* cart, u8 command, bool reply, u8 numparams, ...)
                                                                             ^
1 warning generated.
device_sc64.cpp:108:22: warning: passing an object that undergoes default argument promotion to 'va_start' has undefined behavior [-Wvarargs]
    va_start(params, numparams);
                     ^
device_sc64.cpp:89:152: note: parameter of type 'u8' (aka 'unsigned char') is declared here
  ...cart, u8 command, u8* write_buffer, u32 write_length, u8* read_buffer, u32 read_length, bool reply, u8 numparams, ...)
                                                                                                            ^
1 warning generated.
```

Looking at [some answers](https://stackoverflow.com/q/28054194) on [the internet](https://lkml.org/lkml/2018/10/10/55), it looks like variadic arguments that have their size promoted can be undefined behaviour in the C standard. I was able to resolve these warnings by changing `numparams` from `u8` to `u32`. eg:
```c
void device_sendcmd_sc64(ftdi_context_t* cart, u8 command, u8* write_buffer, u32 write_length, u8* read_buffer, u32 read_length, bool reply, u32 numparams, ...);
// ...
void device_sendcmd_64drive(ftdi_context_t* cart, u8 command, bool reply, u32 numparams, ...);
```

I understand this might not be an immediate issue, but having UB ironed out might be a good idea when it comes to debugging this kind of hardware. I'm unable to test for SummerCart64 or 64Drive though, so I wanted to get thoughts from other people before putting out a PR.

danbolt, 2021-02-13:
If it's not a big deal to change, I can put up a PR later.

danbolt, 2021-02-13:
I put up https://github.com/buu342/N64-UNFLoader/pull/48 for this issue.

@buu342 and @Polprzewodnikowy might be best to decide if this is a good idea or not since I can't test the hardware. Thanks for taking the time to read!

buu342, 2021-02-14:
Putting my comment from the PR here as well:
I don't see any reason as to why `u32` would mess anything up, because I don't think that value is ever sent to the flashcart (it's only so that the code can decide what data to append). I'll check tomorrow morning and, if everything's gucci, I'll merge.

buu342, 2021-02-14:
Merged in 552c10f

closed 2021-02-14 by buu342