buu342/N64-UNFLoader #60: Add Emulator checks to the USB library to prevent crashes on emulators

buu342, 2021-04-27:
Due to the EverDrive check requiring that a register is written to ([which is by design and unlikely to change](https://twitter.com/krikzz/status/1318309612593045505)), it can lead to emulators crashing. A check for whether the ROM is running on an emulator should therefore be added to the start of `usb_initialize()` to prevent further issues. 

This can be done in multiple ways:
1) On emulators, the RDP clock register is always zero (`IO_READ(DPC_CLOCK_REG)`) unless written to specifically by a game.
2) You can check how often the counter register increments, but that fails on MUPEN.
3) You can purposefully trigger an icache miss and measure the counter to see if the CPU stalled, but that requires very specific address alignment
4) Overwrite the start of a function right before it executes without any cache invalidation. This is the method used by Marshall on his MGC 2011 demo.

It would be probably best to attempt method 1, and then fall back on a second method just in case.

tristanpenman, 2021-08-28:
Just to offer a data point, I've implemented method 1) in one of my projects and it appears to work in the following emulators:
- mupen64plus (via mupen64plus-gui, Windows)
- Project64 (Windows)
- sixtyforce (Mac)
- CEN64 (Mac)

Relevant code is here: https://github.com/tristanpenman/asteroids/blob/6332cee87d4882bc82f6d6b73c8a7d3bee3b4abc/n64/debug.c#L251

buu342, 2023-03-26:
Added two emulator checks in [33799ca](https://github.com/buu342/N64-UNFLoader/commit/33799caf11d64eab9522809756734c3b2858cc23).

The first check is the trivial RDP clock register check.
The second check is a VI mirror trick which was suggested to me by Bigbass:
>It turns out that most (maybe all?) RCP registers are mirrored over huge chunks of the memory space. But they are mirrored at different intervals depending on the group. VI for example, mirrors every 0x40 bytes. This means that if you set the VI_CTRL register to something, for example, which is at 0x04400000, you should readback the same value at 0x04400040.
>https://discord.com/channels/205520502922543113/739342626590818335/1042608063191973948 This person apparently did some testing to figure out the mirroring intervals (although I don't know what DPC or DPS registers are, or how big of a range they tested).

The point of these checks is simply to prevent crappy emulators from crashing from the EverDrive check. The emulator check is **opt-in**.

closed 2023-03-26 by buu342