buu342/N64-UNFLoader #63: Possible memory alignment issue in usb_everdrive_writereg

tristanpenman, 2021-07-24:
While integrating the usb and debug examples into my own N64 project, I noticed that the debug library would sometimes fail in various interesting ways, but only when optimisations were enabled `-O1` and `-O2`. I never observed any issues when using `-O0`. For context, I'm using Crash's modern SDK, so this is only relevant for the libultra variants of these functions.

Although this issue was not critical, since my debug builds are compiled at `-O0` anyway, I figured it was worth investigating further. What I eventually found is that `usb_everdrive_writereg`, which calls `usb_everdrive_writedata`, may pass a pointer to a value that is not aligned to an 8 byte boundary. This pointer is then passed to `osPiRawStartDma` or `osPiStartDma`, causing the system to crash.

I've pushed a commit to my fork of N64-UNFLoader that shows what I changed to get this working in my project:
https://github.com/tristanpenman/N64-UNFLoader/commit/b31d69d94e2fc9c0e8e74a295560288548559aa4

I don't want to presume that this fix is actually 'correct' until you've actually looked at it. But if you think this is actually an issue, and that the fix makes sense, I'm happy to raise a PR to make the changes across all of the examples as well.

buu342, 2021-07-24:
Ohh, that's an interesting catch! I don't remember running into issues when I tested it on my end, but alignment never hurts to do! Feel free to open a PR.

tristanpenman, 2021-07-25:
It was a weird one. It only happened on some builds, so it is possible that I'm just doing something unusual that triggers it more often.

In any case, I've raised a PR, so no need to keep this issue open.

closed 2021-07-25 by tristanpenman