buu342/N64-UNFLoader #85: Add a timeout to the EverDrive side of the library

buu342, 2022-07-20:
anacierdem added 24d404e to the library, and it seemed to work fine with the samples, however a user was having trouble where his homebrew was failing to print. Later, we isolated a single sample which presented the same issue, being nu5 from libultra. We tried tweaking the function but we could not get it to work, so we undid the commit. See issue #83 for more information.

So unfortunately the solution proposed in the commit might not work well in a practical sense. Some alternative solution is needed, that is if the risk of the USB stalling permanantly is possible.

buu342, 2022-07-21:
Paging @rasky and @anacierdem

anacierdem, 2022-07-21:
Found the [original reason](https://github.com/krikzz/ED64/blob/d02aef0a57211cf57f8b01b363c41e95207f6846/ED64-XIO/src/bios.c#L154) why this was added.

anacierdem, 2022-07-21:
After @rasky's reminder I remember this was to fix N64 side waiting too long if the PC is not listening and emptying the usb buffer. Also probably it was not waiting indefinitely but was slowing down the code too much. I will check how exactly it behaves.

buu342, 2022-07-21:
That's a perfectly valid reason, the function just needs to be modified to output either error (on timeout) or zero on success, and subsequent calls to the function need to be modified to take the return value into account and branch accordingly, to prevent pushing data to the USB when it is still busy :smile:

buu342, 2023-03-26:
Fixed in [b716f17](https://github.com/buu342/N64-UNFLoader/commit/b716f17321222e772579c3ee29cda6f3a2ca962f)

closed 2023-03-26 by buu342