buu342/N64-UNFLoader #86: __attribute__((aligned(x))); isn't supported by old compilers

burnsba, 2022-08-26:
This is not relevant for gcc and family, only applies when using a very very old compiler.

Lines such as
```
u32 status __attribute__((aligned(8)));
```
cause compiler error. Simply removing the attribute it not sufficient, as the alignment is required by `osPiStartDma` (from docs: "The RDRAM virtual address (vAddr) must be 8-byte aligned"), and mis-aligned address *will* introduce errors (as I have learned).

I considered a solution like
```
    #ifdef __GNUC__
    u32 status __attribute__((aligned(8)));
    #else
    // something else
    #endif
```

but unfortunately the options for "something else" do not really work as a single declaration. Here is how I modified `usb_findcart` (and other methods). My solution is to declare an array on the stack large enough to get an aligned memory address, then declare a pointer to find the first aligned address (using `OS_DCACHE_ROUNDUP_ADDR`). So this will change the existing value previously declared from a value type to a pointer, which alters subsequent method calls.

```
static void usb_findcart()
{
    // PI requires 8 byte aligned address.
    u32 pibuff[5];
    u32 *up;

    up = (u32*)OS_DCACHE_ROUNDUP_ADDR(&pibuff[0]);

    ...

    usb_everdrive_writereg(ED_REG_KEY, ED_REGKEY);
    usb_everdrive_readreg(ED_REG_VERSION, up);

    usb_everdrive_read_version = *up;

    // Check if we have an EverDrive
    if (*up == ED7_VERSION || *up == ED3_VERSION)
    ...
```

buu342, 2022-08-28:
Curious as to the compiler in question. Does it have any documentation, which could point to alternatives?

Otherwise, yeah a memory pool and subsequently trying to figure out what address is aligned seems like the only option we have to solve this.

burnsba, 2022-08-28:
IDO recomp 5.3

buu342, 2022-08-28:
Ah, that's the IRIX compiler. I'll have to see if I can find a manual for it...

buu342, 2023-03-28:
Hey there, I apologize for the wait but I've been going through the issues on this repository this week and been trying to clean them up.

I have removed all uses of `__attribute__((aligned(8)))` in favor of `OS_DCACHE_ROUNDUP_ADDR` (I opted for this over sticking lots of `#ifdef`'s everywhere), and have pushed a fix in [1ccc16c](https://github.com/buu342/N64-UNFLoader/commit/1ccc16cb6616f6d0d07297c84c05ae33f7a672a0). Could you confirm that everything is working as expected on IDO?

closed 2023-04-10 by buu342