buu342/N64-UNFLoader #93: Linux+SC64 udev rules

Dragorn421, 2023-03-05:
after a lot of reading I got the udev rules right (I think) for fixing having to sudo UNFLoader and unbind the default driver:
```sh
$ cat /etc/udev/rules.d/30-sc64.rules 
# SC64 https://github.com/Polprzewodnikowy/SummerCart64
# SC64 - an open source Nintendo 64 flashcart

ATTRS{serial}=="SC64xxxxxx", OWNER="dragorn421"
ACTION=="bind", ATTRS{serial}=="SC64xxxxxx", RUN{program}+="/bin/bash -c 'echo $kernel > /sys/bus/usb/drivers/ftdi_sio/unbind'"
```

----------------------

`ATTRS{serial}` for other flashcarts (or you may use some other way to filter it) may be obtained with e.g.
```
$ udevadm monitor # then plug in the flashcart
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[7641.601052] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1 (usb)
KERNEL[7641.603205] add      /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0 (usb)
...
```
then
```
$ udevadm info -a --path /devices/pci0000:00/0000:00:14.0/usb3/3-1
    ATTR{manufacturer}=="Polprzewodnikowy"
    ATTR{maxchild}=="0"
...
    ATTR{product}=="SC64"
    ATTR{quirks}=="0x0"
    ATTR{removable}=="removable"
    ATTR{rx_lanes}=="1"
    ATTR{serial}=="SC64xxxxxx"
```

----------

sources:
`man udev`
`man udevadm`
<https://stackoverflow.com/questions/47563963/automatically-bind-a-device-to-a-specific-driver>
<https://unix.stackexchange.com/questions/78169/how-can-i-prevent-drivers-from-binding-usb-devices-on-a-certain-bus>
<https://askubuntu.com/questions/1021547/writing-udev-rule-for-usb-device>

Dragorn421, 2023-03-06:
Note these changes (just like `rmmod ftdi_sio` or unbinding the kernel's default ftdi driver from the sc64) prevent the "official" sc64 tools from working as they rely on the /dev/ttyUSBX provided by the default driver

If someone ever makes a script to easily bind/unbind ftdi_sio to the device please share (atm my low-complexity approach is "comment out udev rules and plug out device and back in")

buu342, 2023-04-09:
Added in a003602. You can now use `installer_linux.sh` to set up UNFLoader for you.

closed 2023-04-09 by buu342