buu342/N64-UNFLoader #94: `device_sendrom` lacking error checking

Dragorn421, 2023-03-05:
## Expected Behavior
Calling `device_sendrom("/")` should error in some way, since `/` is not a file

## Current Behavior
It does not error and happily tries to upload a ROM. In the case of the SC64, it eventually overwrites the bootloader because of an unrelated issue

## Possible Solution
Add more error checking. I'll open a PR

## Steps to Reproduce
beware my jank python bindings:
```c
extern "C" {
    void  c_device_find(int automode) {device_find(automode);}
    void  c_device_set_64drive1(ftdi_context_t* cart, int index) {device_set_64drive1(cart, index);}
    void  c_device_set_64drive2(ftdi_context_t* cart, int index) {device_set_64drive2(cart, index);}
    void  c_device_set_everdrive(ftdi_context_t* cart, int index) {device_set_everdrive(cart, index);}
    void  c_device_set_sc64(ftdi_context_t* cart, int index) {device_set_sc64(cart, index);}
    void  c_device_open() {device_open();}
    void  c_device_sendrom(char* rompath) {device_sendrom(rompath);}
    bool  c_device_testdebug() {return device_testdebug();}
    void  c_device_senddata(int datatype, char* data, u32 size) {device_senddata(datatype, data, size);}
    bool  c_device_isopen() {return device_isopen();}
    DWORD c_device_getcarttype() {return device_getcarttype();}
    void  c_device_close() {device_close();}
}
```
```diff
$ git diff UNFLoader/Makefile
diff --git a/UNFLoader/Makefile b/UNFLoader/Makefile
index 1d40100..6219354 100644
--- a/UNFLoader/Makefile
+++ b/UNFLoader/Makefile
@@ -1,4 +1,4 @@
-APP=UNFLoader
+APP=UNFLoader.so
 OS_NAME := $(shell uname -s)
 
 CODEFILES=main.cpp helper.cpp device.cpp debug.cpp \
@@ -14,11 +14,12 @@ CXX=g++
 ifeq ($(OS_NAME),Darwin)
        DEPENDENCIES := -lncurses -lftd2xx -lpthread
 else
-       DEPENDENCIES := -lncurses -lftd2xx -lpthread -lrt 
+#      DEPENDENCIES := -lncurses -lftd2xx -lpthread -lrt 
+       DEPENDENCIES := -lncurses -L /home/dragorn421/Documents/n64homebrew/libftd2xx-x86_64-1.4.27-release/build/ -lftd2xx -lpthread -lrt 
 endif
 
-LINKER_OPTIONS := -Wl,-rpath /usr/local/lib
-CFLAGS=-D LINUX -D_XOPEN_SOURCE_EXTENDED
+LINKER_OPTIONS := -Wl,-rpath /usr/local/lib -shared
+CFLAGS=-D LINUX -D_XOPEN_SOURCE_EXTENDED -fPIC
 
 default: $(APP)
```
```py
from pathlib import Path
import ctypes

cdll = ctypes.CDLL(Path(__file__).parent / "UNFLoader.so")
# cannot dynamically load position-independent executable

print("c_device_find")
cdll.c_device_find(0)
print("c_device_open")
cdll.c_device_open()
print("c_device_sendrom")
cdll.c_device_sendrom("/home/dragorn421/Documents/n64homebrew/echasses/ctest.z64")
#cdll.c_device_sendrom("/home/dragorn421/Documents/n64homebrew/ctrlpak_dump/romfile.z64")

print("end")
```

```diff
diff --git a/UNFLoader/device.cpp b/UNFLoader/device.cpp
index df75d6a..20b68c6 100644
--- a/UNFLoader/device.cpp
+++ b/UNFLoader/device.cpp
@@ -226,6 +226,8 @@ void device_sendrom(char* rompath)
     time_t lastmod = 0;
     struct stat finfo;
 
+    printf("device_sendrom rompath = %s\n", rompath);
+
     for ( ; ; )
diff --git a/UNFLoader/device_sc64.cpp b/UNFLoader/device_sc64.cpp
index 70637ca..8ffc520 100644
--- a/UNFLoader/device_sc64.cpp
+++ b/UNFLoader/device_sc64.cpp
@@ -195,6 +195,8 @@ void device_sendrom_sc64(ftdi_context_t* cart, FILE* file, u32 size)
         "SRAM banked 3x32k",
     };
 
+    printf("device_sendrom_sc64 size = %d = 0x%X\n", size, size);
+
     // 256 kB chunk size
     chunk = 256 * 1024;
```
```
device_sendrom rompath = /
device_sendrom_sc64 size = -1 = 0xFFFFFFFF
```

## Context (Environment)
SC64, Kubuntu Linux, python bindings

## Detailed Description
I'll be opening a PR

## Possible Implementation
I'll be opening a PR

buu342, 2023-03-05:
Merged your PR, but I'll keep this open as a reminder for my current rewrite. Thank you :smile:

Dragorn421, 2023-03-05:
whoopsie ;)

buu342, 2023-03-13:
Fixed in 4a8291e. I'll close this issue when the rewrite branch gets merged.

closed 2023-04-10 by buu342