buu342/N64-UNFLoader #112: `usb_everdrive_poll` sometimes fails to extract the correct packet size

Thar0, 2023-06-14:
## Expected Behavior
`usb_everdrive_poll` should always compute the correct size of the inbound packet.

## Current Behavior
`usb_everdrive_poll` sometimes computes an incorrect packet size based on the size of the inbound packet and compiler flags/configuration. See below for details.

## Possible Solution
Use `unsigned char` for greater compiler portability.
https://github.com/Thar0/N64-UNFLoader/commit/c3d562ed4df85790cbaab2f1f2d93dc4edf8e883

## Steps to Reproduce
1. Use a supported Everdrive model (I used an Everdrive X7)
2. On N64 wait for a packet to arrive, e.g.
```c
    unsigned long header = 0;
    while (header == 0)
        header = usb_poll();
```
3. Send a packet over USB where either the second or third byte of the packet size has the msbit set (e.g. packets sized 0x80, 0xF0, 0x180)
4. `usb_poll` will continue to return 0 instead of a valid header for such packets, while packets without the msbit set in either of these bytes (e.g. packet sized 0x40) will arrive with no problems.

## Context (Environment)
I wanted to send packets to a program running on the N64 via Everdrive USB and stumbled upon this bug while using the usb library as part of libdragon. The GCC build in the libdragon toolchain uses signed char, as GCC defaults to signed char if not configured otherwise. This invites the bug described below.

## Detailed Description
On compilers where `char` defaults to signed, the following line will produce wrong results due to sign extension:
`usb_datasize = (int)buff[5]<<16 | (int)buff[6]<<8 | (int)buff[7]<<0;`
For example, a seemingly-innocuous intended packet size of 0x000080 becomes 0xFFFF80, as 0x80 is sign extended to 0xFFFFFF80.

## Possible Implementation
https://github.com/Thar0/N64-UNFLoader/commit/c3d562ed4df85790cbaab2f1f2d93dc4edf8e883

buu342, 2023-06-15:
I'm really confused about this one. Why would the sign produce different results in the line `usb_datasize = (int)buff[5]<<16 | (int)buff[6]<<8 | (int)buff[7]<<0;`? The sign of `buff` shouldn't make a difference to the data contained within.

buu342, 2023-06-15:
Should be fixed in [5c6c016](https://github.com/buu342/N64-UNFLoader/commit/5c6c01638b9dc53e76a84c1600391e6f780b7df5).