buu342/N64-UNFLoader #130: UNFLoader sometimes crashes on ROM upload or when debugging

mwpenny, 2025-01-09:
## Context (Environment)

I am using Libultra + EverDrive X7 (OS version 3.08). I am trying to upload a ROM (~24 MB) to debug with GDB.

## Steps to Reproduce

1. Create/obtain a non-small ROM so that it will take some time to upload to a flashcart
2. Connect PC to flashcart
3. Use `UNFLoader -r <rom file>`
4. Use `UNFLoader -d -g`
5. Connect GDB following the instructions in [USB+Debug Library/README.md](https://github.com/buu342/N64-UNFLoader/blob/master/USB%2BDebug%20Library/README.md).

## Expected Behavior

* The ROM upload is successful
* GDB connection is successful

## Current Behavior

* Sometimes UNFLoader crashes during upload
* Sometimes UNFLoader crashes when connecting GDB

## Possible Solution

I debugged this and it is crashing because multiple threads are reading/modifying the terminal and debug `local_mesgqueue`s at once.

I fixed this using mutexes.

buu342, 2025-01-10:
This should be fixed with the merge of PR #132, please confirm :slightly_smiling_face: 

One thing I'm not quite understanding is how the `local_mesgqueue` is causing the ROM upload to sometimes crash. The message queue shouldn't be touched during the upload process, unless you are reuploading a ROM while another is currently running.

mwpenny, 2025-01-10:
No more crash. Thanks!

> One thing I'm not quite understanding is how the race condition on local_mesgqueue is causing the ROM upload to sometimes crash.

During ROM upload, if using ncurses (default), `termthread()` is constantly checking for new messages in `local_mesgqueue` in order to print them to the terminal. At the same time, `__log_output()` could be called by another thread that wants to do some logging. For example, `progressthread()` will update the upload progress bar.

Modifying `local_mesgqueue` (`std::queue`) is not an atomic operation, and so for instance a push/pop at the same time from different threads can corrupt the queue state.

ROM uploading is an easy place where this can happen, but the issue could happen any time something is logged by a thread other than the terminal thread. The debug `local_mesgqueue` had a similar problem, but it related to queuing/sending GDB packets rather than terminal output (similar code).

closed 2025-01-10 by mwpenny

buu342, 2025-01-10:
Oh pfft, doink! Yeah of course, I completely forgot about the ncurses thread lmao.