buu342/N64-UNFLoader #131: Invalid GDB responses when using Libultra

mwpenny, 2025-01-09:
## Context (Environment)

I am using Libultra + EverDrive X7 (OS version 3.08). I am trying to debug with GDB. I have defined `USE_RDBTHREAD`.

## Steps to Reproduce

1. Connect PC to flashcart
2. Load ROM with UNFLoader debug library enabled
3. Use `UNFLoader -d -g`
4. Connect GDB following the instructions in [USB+Debug Library/README.md](https://github.com/buu342/N64-UNFLoader/blob/master/USB%2BDebug%20Library/README.md).

## Expected Behavior

GDB connection is successful

## Current Behavior

The exact error can differ from run to run. Here are some I have seen:
* UNFLoader: `Error: Unexpected DMA header.`
* UNFLoader: `Error: Received bad CMP signal.`
* GDB: `Truncated register 65 in remote 'g' packet`
* GDB: `Remote 'g' packet reply is of odd length`

Sometimes in the UNFLoader output I can see contents from multiple messages being sent back as one (note that this contains "DMA@" within it):
```
Replying with '0*,f*+e0**29f*"ff800e92780*+10*(3242f*'ef* 0*,0*(ff010*(ff00f*"ff800f0* 0*(ff000*,0*,0*,0*,f*"ff800aca9c0*+10**120*+10*+1f*"ff800f0* f*"ff800acaa00*+80*,0*,x*,x*,0*,f*"ff800a42500*,f*"ff8003abb00*(ff030*,0*,0DMA@☺0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*,0*%10008000*&800b11'
```

## Possible Solution

The debugger code relies on the USB thread having the highest priority so other threads don't interrupt it during reads/writes. However, it is possible for the USB thread to block in `usb_dma_write()` when waiting for the DMA to complete. During this time, another thread (such as the RDB thread) can call USB related functions and corrupt the internal state.

I fixed this using the same workaround as in `usb_dma_read()`: use a busy loop to wait for the DMA to complete. This could also be solved using `osSendMesg()` / `osRecvMesg()` but I went with the existing method.

mwpenny, 2025-01-09:
Also - not sure if this is due to my setup or a known issue. After hitting a breakpoint I can't hit it again unless I remove/re-add it. I see GDB sending a `z` packet when I continue (restoring the original instruction) but it doesn't get re-added.

Let me know if I should open a separate issue for this. I'm using `gdb-multiarch` version 14.1.

buu342, 2025-01-10:
This should be fixed with the merge of PR #132, please confirm :slightly_smiling_face:

mwpenny, 2025-01-10:
I realized I missed one thing: in `debug.c`, `debug_rdb_qsupported()` [does not copy the null terminator](https://github.com/buu342/N64-UNFLoader/blob/725a07ae34b870fe04f6116c483cfb82ea87e8b1/USB%2BDebug%20Library/debug.c#L1439) to the response.

Depending on the previous buffer contents, sometimes everything is fine, but other times breakpoints don't work since GDB doesn't know the target supports software breakpoints (invalid qSupported response).

I noticed the other responses are null terminated so fixing this is straightforward. I've opened #133. Sorry for the double PR.

buu342, 2025-01-10:
Not an issue, it's appreciated!

closed 2025-01-10 by mwpenny