buu342/N64-UNFLoader #138: Minor issues with macOS builds

command-tab, 2025-08-15:
Hello! Just noting some minor issues with the macOS build:

- UNFLoader releases are named with `x64` like Intel binaries but are in fact `arm64` Apple Silicon binaries. (I don't know if a Universal binary is desired since x86-64 macOS isn't long for this world, though it is still officially supported until around this time next year.)
- The macOS binary should be made executable (`chmod +x UNFLoader`) before archiving, so it's executable upon un-archiving without manual intervention.

Neither is a showstopper, but the first is misleading and the second saves users a step and possible confusion. Happy to open a PR if that's of interest!

buu342, 2025-08-16:
Thanks for the suggestions. I'm currently battling CI build nonsense on Windows, so I'll make a note to tweak the Apple binaries + zips

buu342, 2025-08-16:
Should be fixed by now in the CI.

command-tab, 2025-08-20:
The most recent build from Azure has the same issue -- binary is correctly present, but did not retain permissions.